RaspberryPiFoundation/picozero #47: Add a Motor device

martinohanlon, 2022-04-27:
Add a `Motor` device to picozero to support single motors which are driven by a motor controller.

Initial thoughts on constructor methods etc.

Constructor

```python
__init__(self, forward_pin, backward_pin, pwm=True)
```

Methods

```python
.forward(speed=1)
.backward(speed=1)
.right(speed=1)
.left(speed=1)
.stop()
```

Properties

```python
.value # -1 to 1 
.is_active # True if the motor is moving
```

tracygardner, 2022-04-27:
Maybe I'm confused but they sound more like methods for a vehicle with two motors? I don't tend to think of a single motor in forward/backward or left/right terms. Clockwise and anti/counter clockwise? I've always liked the Scratch/LEGO this way/that way. Let's make sure that it's intuitive to build a device like a fan, windmill, turntable or pulley. 

An option would be this style of API: https://en.scratch-wiki.info/wiki/Motor_Blocks. Direction (clockwise anti/counter-clockwise) could be included in the on method. Having methods that run the motor for a number of seconds is great for beginners (and most beginner APIs take this approach.) You would want an option to keep the motor running in a direction until you stop. A `reverse` method is also useful. Also it's useful to have a default direction when you don't care which way the motor goes, you just want it to go.

martinohanlon, 2022-07-22:
Added first prototype for `Motor` to the motor branch. https://github.com/RaspberryPiFoundation/picozero/blob/motor/picozero/picozero.py#L1114

I went with `forward` `backward` for the move methods. Suggest we gather opinions as alias as appropriate.

Added `t` and `wait` to `forward` and `backward` so you can specify how long it should move and whether it should wait before returning.

e.g.

~~~python
    def forward(self, speed=1, t=None, wait=False):
    def backward(self, speed=1, t=None, wait=False):
~~~

martinohanlon, 2022-07-27:
Im thinking `Motor` need an `on`  or `move` method as to @tracygardner 's comment.

> Also it's useful to have a default direction when you don't care which way the motor goes, you just want it to go.

`on` / `move` parameters?

```python
def move(self, speed=1, t=None, wait=False)
```

`speed` would accept values between `-1` and `1` - moving either "forward" or "backward"

martinohanlon, 2022-07-27:
Added a `move` method and created an alias between `move` and `on` and `stop` and `off`.

martinohanlon, 2022-07-27:
Also added tests

martinohanlon, 2022-07-27:
Fyi - as a coincidence of adding `t` and `wait` to PWMOutputDevice (so we can control motors in that way) you can now also do similar with the `on` method for other output devices e.g. LEDs, Buzzers, etc.

e.g. if you wanted to turn an LED on for 2 seconds and then have it automatically turn off you would use.

```python
l = LED(1)
l.on(t=2)
```

martinohanlon, 2022-08-05:
Feedback that `move` should be renamed as `start`. That makes sense to me.

martinohanlon, 2022-08-13:
Release with v0.3.0

closed 2022-08-13 by martinohanlon

tibaultbordeaux, 2023-05-24:
After renaming from `move` to `start` one should adjust the Recipe in the docs.
[picozero.readthedocs.io](https://picozero.readthedocs.io/en/latest/recipes.html#motor)

Once again, thank you for this great work!