RaspberryPiFoundation/picozero #49: Add a Robot

martinohanlon, 2022-04-27:
Add a Robot class which allows users to control 2 motors on a typical "robot buggy" which has 2 rear wheels with a third wheel which pivots. 

Thoughts on constructor, methods and properties.

Constructor

```python
def __init__(self, left, right, pwm=True)
# left and right are a tuple of 2 pins e.g. left=(1,2)
```

Methods

```python
.forward(speed=1)
.backward(speed=1)
.left(speed=1)
.right(speed=1)
.stop()
```

Properties

```python
.value # returns a tuple of values from -1 to 1 representing the value of each motor e.g. (1,1)
```

tracygardner, 2022-04-27:
Maybe call this Buggy rather than Robot if that's what it is? I tend to think servos when I think Robot and there are lots of interesting use cases for that sort of robot.

Again, adding methods with a duration is helpful for beginners. 

Might be useful to add a balance factor for the case where motors aren't equally balanced to make it easier to use cheap motors? (Yes, I've spent many hours with buggies that veer off in one direction!)

martinohanlon, 2022-07-28:
Added a `Robot` class with an alias of `Rover`. Rover seemed to be more accepted in the very unscientific research I did via twitter.

Changes are currently in the `motor` branch - https://github.com/RaspberryPiFoundation/picozero/tree/motor

martinohanlon, 2022-08-13:
Release with v0.3.0

closed 2022-08-13 by martinohanlon