RaspberryPiFoundation/picozero #58: Error using GP9

topshed, 2022-06-05:
**Describe the bug**
Using GP9 with an LED raises an error

**To Reproduce**

```python
from picozero import LED

l = LED(9)

l.blink()
```
leads to the Error:

> Traceback (most recent call last):
>   File "<stdin>", line 3, in <module>
>   File "/lib/picozero/picozero.py", line 559, in LED
>   File "/lib/picozero/picozero.py", line 355, in __init__
>   File "/lib/picozero/picozero.py", line 367, in _check_pwm_channel
> PWMChannelAlreadyInUse: PWM channel 4B is already in use by pin 25. Use a different pin

Note, this *does* work ok:

```python
from machine import Pin
l9 = Pin(9,Pin.OUT)
l9.toggle()
```

**Expected behaviour**
If I use another pin (e.g pin 8, then the LED will blink. 

**System information:**

 - OS: Raspberry Pi OS (Bullseye)
 - Development environment Thonny 3.3.14 
 - picozero version:  0.1.0

martinohanlon, 2022-06-06:
picozero is working as expected...  however that doesnt mean improvements cant be made :)

The context is:
- by default `LED` uses a PWM channel (this way methods like `pulse` and `brightness` are available)
- PWM channels are shared between pins (9 and 25 use the 4B PWM channel)
- pin 25 is used on the onboard LED `picozero.pico_led` 
- micropython wont raise an error if you use the same PWM channel but "weird" things happen - hence picozero raising an error and not letting you do that

This code works because it uses a "digital" pin and not PWM:

```
from machine import Pin
l9 = Pin(9,Pin.OUT)
l9.toggle()
```

You can tell `LED` to not use PWM by using:

```
l9 = LED(9, use_pwm=False)
```

... which wouldnt have raised an error.

Improvements can probably be made to the error message... however this is a general error message for all PWM based devices (LED, Buzzer, Speaker, etc) so we will be constrained regarding how specific it can be made.

@topshed clearly the error message didnt give you enough information - any thoughts on what might have helped you?

topshed, 2022-06-06:
Thanks for the great explanation. I guess it would be helpful  the error could alert the user to the fact that GP9 is (intentionally) going to behave differently to other pins. I'd wired up LEDs to pins 1-10 and it seemed odd that this one pin was not behaving in the same way as the others.

Maybe there is also an issue here about picozero being different to gpiozero, where PWMLED exists separately fro LED?

martinohanlon, 2022-07-04:
This has been resolved with the latest release. The `pico_led` no longer uses a PWMLED so GP9 is available to use as a PWM pin. 

fyi - pico_led was changed because the pico w onboard led cant use a PWM pin, and I thought it better that they were consistent.

closed 2022-07-04 by martinohanlon