RaspberryPiFoundation/picozero #67: Remote Pins (picod)

bsimmo, 2022-07-27:
I know it is early in your development, but in a similar way that gpiozero on the Pi can work remotely* with pigpio pin factory, E.g activate a gpio or motor from another Pi or PC. 

In future it would be nice to do similar on the PiPico using the picod daemon. https://abyz.me.uk/picod/py_picod.html

* Remote GPIO with gpiozero for anyone reading that doesn't know https://gpiozero.readthedocs.io/en/stable/remote_gpio.html
I thought I would just put it out there as a future feature since this request popped up on the Pi Forum.

martinohanlon, 2022-07-27:
I dont see this becoming a priority in the short or medium term. I would also question whether it fits with picozero aims of being lightweight, however time will tell.

I think a minimum pre-requisite for this would be a micropython module which supports remote access (which picozero can build upon rather than implementing it into picozero).

A key technical constraint is that picozero is a micropython module (i.e. its not a Python and wont run on a "computer") and its designed run directly on pico built using functions like timers and interupts which are hardware dependent.

bsimmo, 2022-12-21:
Sorry for not getting back.
I think I miss-understood this module back then nor appreciate the limitation of memory (especially with the W variant now.)

It's a shame micropython cannot be run on a 'computer' Pi/Win or otherwise like circuitpython can, but anyway.

I guess what I was thinking is a Python3* version of _picozero_ to run on a Pi/Win via Thonny for example. _picod daemon_ running on a Pico to enable remote gpio access and then just write the program using your simplified picozero api and talk to any number of Pico's connected .  
Not something for this version of picozero :-)
As for controlling other pico's gpio from a pico.. via picozero. that may be a bit too much :-)

* or desktop implementation of micropython when I first wrote the idea.