RaspberryPiFoundation/picozero #78: Potentiometer-pwmled example should be added

sentairanger, 2022-10-15:
Hello, I was trying to find a way to use a potentiometer to change the value of an LED and I was able to get it to work. I created this issue to recommend adding this sample code I created to the official documentation as an example for others to try out. Thank you. Sample code below works as expected.

```python
from picozero import Pot, PWMLED
from time import sleep

pot = Pot(0)
red =  PWMLED(16)

while True:
    print(pot.value, pot.voltage)
    red.on(value=pot.value)
    sleep(0.1)```

martinohanlon, 2022-10-19:
Thank you for your suggestion. I will add it to the recipes the next time I do a release.

sentairanger, 2022-10-19:
Thank you Martin I will now close this issue.

martinohanlon, 2022-10-19:
I have reopened the issue as the change hasn't yet been made.

martinohanlon, 2022-11-18:
I have added a recipe to the documentation.

closed 2022-11-18 by martinohanlon