RaspberryPiFoundation/picozero #91: DigitalInputDevice.when_activated method has incorrect decorator

liminspace, 2022-12-20:
Please, fix the bug in the class DigitalInputDevice:

this
```python
    @when_activated.setter
    def when_deactivated(self, value):
        self._when_deactivated = value
```
must be
```python
    @when_deactivated.setter
    def when_deactivated(self, value):
        self._when_deactivated = value
```

martinohanlon, 2022-12-20:
Thank you. Looks like a classic cut and paste error. Its not been noted before.

Did you experience an error or issue or did you notice the issue in the code?

I am interested because I just tested this program

```python
from picozero import DigitalInputDevice

b = DigitalInputDevice(7, pull_up=True)

def pressed():
    print("pressed")

def released():
    print("released")

b.when_activated = pressed
b.when_deactivated = released
```

and it works as expected?! Is there a difference here between Python and Micropython?

martinohanlon, 2022-12-20:
Fyi - I'll do a release later this week.

liminspace, 2022-12-20:
I had some issue with it, but don't remember details. I had to make this changes to make it works properly.

Also I can propose supporting brightness for RGBLED.
I implemented it as possible fourth item of a color value.

```python
class RGBLED(OutputDevice, PinsMixin):
    def __init__(
        self, red=None, green=None, blue=None, active_high=True, initial_value=(0, 0, 0), pwm=True
    ):
        ...
        super().__init__(active_high, self._apply_brightness(initial_value))

    def _apply_brightness(self, value):
        if isinstance(value, tuple) and len(value) == 4:
            brightness = value[3]
            value = tuple(v * brightness for v in value[:3])
        return value

    @value.setter
    def value(self, value):
        self._stop_change()
        self._write(self._apply_brightness(value))

    @color.setter
    def color(self, value):
        if len(value) == 4:
            self.value = tuple(self._from_255(v) for v in value[:3]) + (value[3],)
        else:
            self.value = tuple(self._from_255(v) for v in value)
```

for example:
```python
rgbled.color = (255, 200, 50, 0.5)
```
The range of brightness must be 0.0...1.0.
I use it in my project and I see it to be quite useful.
It can be implemented better, like using some logarithmic curve for percetage/brightness, but for most cases it's not so important.

martinohanlon, 2022-12-20:
>I had some issue with it, but don't remember details. I had to make this changes to make it works properly.

Ok. 

Can you raise a seperate issue for new feature requests pls. Although my first thought is that adding `brightness` to RGBLED adding functionality which can already be acheived by lowering the intensity of the RGB values and given the key principle of being lightweight I dont feel this would be a priority.

martinohanlon, 2022-12-20:
fix is in `dev` branch

martinohanlon, 2022-12-22:
Resolved in v0.4.1

closed 2022-12-22 by martinohanlon