RaspberryPiFoundation/picozero #97: blink_onboard_led.py example code has errors

mcp2319, 2023-01-20:
**Describe the bug**

The example code in "blink_onboard_led.py" has errors and does not run.
(Bug #1) line 4 "from pico" needs to be changed to "from picozero".
(Bug #2) line 11 calls funtion "led.blink_stop() which does not seem to be defined in the library.  Changing this line to "led.blink(0)" is how I fixed it.  Hope this help.  I love this library for the pico and RP.  Thank you for making it!

```python
01 # Blink the onboard LED, sleep for 10 seconds then stop the blinking
02 
03 from time import sleep
04 from pico import Button, LED
05 
06 led = LED()
07
08 led.blink(0.5)
09 print("Blinking")
10 sleep(10)
11 led.blink_stop()
12 print("Stopped blinking")
```

martinohanlon, 2023-01-23:
Thanks for the report 

The examples in the /examples folder were created during the development process and have clearly not been updated.

They have now been superceded by the examples in /docs/examples

I would suggest we remove them and provide a link from the README and docs to the new code examples.

 @mcp2319 the "typical" way of stopping the blink would be to call led.off()

martinohanlon, 2023-02-02:
I have removed the old examples and provided a link in the README to /docs/examples.

closed 2023-02-02 by martinohanlon