RaspberryPiFoundation/picozero #99: Add MotionSensor

sentairanger, 2023-02-17:
I would like to see MotionSensor (PIR sensor) added to picozero similar to MotionSensor on gpiozero.

martinohanlon, 2023-02-23:
Based on `picozero.DigitalInputDevice` I think the implementation would look like this.

```python
class MotionSensor(DigitalInputDevice):
    def __init__(self, pin, pull_up=False, bounce_time=0.02): 
        super().__init__(pin=pin, pull_up=pull_up, bounce_time=bounce_time)

MotionSensor.motion_detected  = MotionSensor.is_active
MotionSensor.when_motion = MotionSensor.when_activated
MotionSensor.when_no_motion = MotionSensor.when_deactivated
```

This implementation would be different to gpiozero. gpiozero uses the concept of a `SmoothedInputDevice` which takes the average of continuous readings to smooth the input, presumably to stop false triggers (although in practice I dont think this really helps, PIRs always being a bit "jittery").

sentairanger, 2023-02-23:
Would it help to add an interrupt request to reduce jitter? I will try this code out and make changes if needed. I'll report back if I can reduce jitter.

martinohanlon, 2023-02-24:
Not sure what you mean by interupt request. DigitalInputDevice already used interupts and implements debouncing -https://github.com/RaspberryPiFoundation/picozero/blob/master/picozero/picozero.py#L1589