RaspberryPiFoundation/picozero #101: Class DigitalInputDevice: Missing factor 1000 in debouncing when recalcualting stop time

SmokeyBrandie, 2023-03-15:
In this line https://github.com/RaspberryPiFoundation/picozero/blob/dev/picozero/picozero.py#L1601
the stop time is resetted, when state changes within debouncing time.

`stop = ticks_ms() + self._bounce_time`

From my point of view there is a missing "* 1000", because "self._bounce_time" is in seconds.

This line should be exactliy like the one above: https://github.com/RaspberryPiFoundation/picozero/blob/dev/picozero/picozero.py#L1597

`stop = ticks_ms() + (self._bounce_time * 1000)`

martinohanlon, 2023-05-03:
Yes, I agree, it just looks like a simple oversight. It would have affected the bounce time when values changed.

I have prepared a fix in PR #104

martinohanlon, 2023-05-15:
This has been resolved in 0.4.2 

@MarcScott - can you close the issue please.

PR #104 (2023-05-11): Resolve issue 101 - digitalinputdevice bounce time
Resolve #101
closed 2023-05-15 by MarcScott