RaspberryPiFoundation/picozero #106: Version 0.4.2

martinohanlon, 2023-05-12:
I am going to prepare a new release of picozero - 0.4.2 - it will include a couple of small bug fixes but I am doing it mainly to run through the release process using a fork of the main repo.

I'll keep track of where we are in the process in this issue and assign tasks which I cant do myself.

martinohanlon, 2023-05-12:
@MarcScott I cant assign you to an issue, hopefully if I tag you, you will see it.

martinohanlon, 2023-05-12:
Create PR #107 for version 0.4.2

@MarcScott could you review please.

martinohanlon, 2023-05-15:
I have released version 0.4.2 to PyPI - https://pypi.org/project/picozero/0.4.2/

@MarcScott can you:
- create a PR and merge the `dev` branch into `main`
- create a release with the tag 0.4.2 and upload the release tar.gz file to it - https://files.pythonhosted.org/packages/f5/b5/8a241bbd029b5116d36070ebc0f13caa6efa3a77cfe174e889418e60b8f4/picozero-0.4.2.tar.gz
- close a couple of issues which have been resolved by the release

martinohanlon, 2023-05-15:
#101 and #103 have been resolved with this release. @MarcScott can you close them please.

callimero, 2023-11-17:
I wonder is this project dead?