RaspberryPiFoundation/picozero #108: Question: Does it work with driver boards and/or other RP2040 boards?

tibaultbordeaux, 2023-05-16:
Dear devs, 

please allow me to ask a question:

Does picozero work with 

a) driver boards like the [Kitronik Robotics Board 5329 (github.com)](https://github.com/KitronikLtd/Kitronik-Pico-Robotics-Board-MicroPython)

b) RP2040 based boards like the [Cytron Maker Pi RP2040 (github)](https://github.com/CytronTechnologies/MAKER-PI-RP2040)

especially the motor, servo and ultrasonic parts?

Thanks in advance and thanks for doing such a great job!

tibaultbordeaux, 2023-05-24:
I just got a Maker Pi RP2040 on my hand and can answer at least question b):
Yes, it just works. 
You can install the original Raspberry Pico firmware on it and it gets recognized by Thonny as a Raspberry Pico.
picozero is doing all the stuff like expected.

Well done!

pjbRPF, 2025-12-11:
Thanks for raising this!

picozero is designed to be board-agnostic, so it works with any RP2040 board as long as you use the correct GPIO pin numbers in (for example) the Motor, Servo, or DistanceSensor classes.
For the Maker Pi RP2040, someone has confirmed above that it “just works” because the motor/servo outputs are driven directly by RP2040 GPIOs.

The Kitronik Robotics Board 5329 should work the same way as its motor and servo headers are ultimately connected to standard Pico GPIO pins, so you only need to map the correct pins into picozero.

Docs for reference:
https://picozero.readthedocs.io/en/latest/api.html

If there’s nothing further needed, I’ll close this issue, but feel free to reopen if you hit any problems.

closed 2025-12-11 by pjbRPF

tibaultbordeaux, 2025-12-11:
Thanks, Pete! That's fantastic!