RaspberryPiFoundation/picozero #119: EventFailedScheduleQueueFull in "/lib/picozero/picozero.py", line 1632, in _pin_change

ericterpstra, 2025-02-27:
**Describe the bug**

I'm trying to use the picozero library. The most simple, basic code given in the docs is not working. The error message received is:

```
Traceback (most recent call last):
  File "/lib/picozero/picozero.py", line 1632, in _pin_change
EventFailedScheduleQueueFull: Button (pin 13) - led_on_off not run due to the micropython schedule being full
```

**To Reproduce**

I attached a simple pushbutton to PIN 13 and GND.  I also tried PIN 18 and a different GND. Same results no matter which pin is used. I was not pressing the button in rapid succession. In fact, pressing the button, then waiting several seconds, then pressing again yielded the same error.

I tried these examples from the picozero docs:

```
from picozero import Button, pico_led
from time import sleep

button = Button(13)

def led_on_off():
    pico_led.on()
    sleep(1)
    pico_led.off()
    sleep(1)
    
button.when_pressed = led_on_off
```

Also

```
from picozero import Button, pico_led

button = Button(13)

button.when_pressed = pico_led.on
button.when_released = pico_led.off
```

Both of these examples cause the error mentioned above _most_ of the times the button is pushed. Once in a while (maybe 1 out of 10 or 15 attempts), the intended behavior occurs.

**Expected behaviour**

This code (without the use of picozero) works _as expected_:

```
from machine import Pin
import time

button = Pin(13, Pin.IN, Pin.PULL_UP)
led = Pin("LED", Pin.OUT)

while True:
    if button.value() == 0:
        led.value(1)
    else:
        led.value(0)
    time.sleep(0.1)
```

**System information:**

 - OS: Raspberry Pi Pico W with Micropython 1.24.1
 - Development environment: Both Thonny and VSCode + MicroPython extension.
 - picozero version: 0.4.1

scruss, 2025-03-12:
There's some discussion here on the forum: [EventFailedScheduleQueueFull on simple Button press with picozero - Raspberry Pi Forums](https://forums.raspberrypi.com/viewtopic.php?t=384683)

While I can't personally replicate the error, it's not working as it should for the OP

pjbRPF, 2025-11-26:
Could not replicate, despite spamming a button!